TypQxQ/Klipper_ToolChanger #18: Unable to load module 'ktcclog'

MarcoFranchi, 2023-02-25:
After update to v0.0.3-7-g3065012 I have this problem:

`Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 130, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/toollock.py", line 469, in load_config
    return ToolLock(config)
  File "/home/pi/klipper/klippy/extras/toollock.py", line 36, in __init__
    self.log = self.printer.load_object(config, 'ktcclog')
  File "/home/pi/klipper/klippy/klippy.py", line 120, in load_object
    raise self.config_error("Unable to load module '%s'" % (section,))
configparser.Error: Unable to load module 'ktcclog'
webhooks client 1965907888: New connection
webhooks client 1965907888: Client info {'program': 'Moonraker', 'version': 'v0.8.0-5-g160f6d6'}
`
any idea for fixing?
thankyou a lot
   MArco

TypQxQ, 2023-02-25:
Should mean that  you have to include the new section [ktcclog] before the [toollock] section in the configuration files :D

TypQxQ, 2023-02-25:
There are a couple of new things in the latest release, I tried to list them at the end of document.

MarcoFranchi, 2023-02-26:
thankyou for your quickly replay.
Yes I have edit tool.cfg as I read in a documentation with the section [ktcclog] but without  success
after re-install from fresh all work fine.. I not able to understand way but the problem is fixed
thankyou for a great tool.
   MArco

TypQxQ, 2023-02-27:
That is strange, but I'm happy it works :)

closed 2023-02-27 by TypQxQ