TypQxQ/Klipper_ToolChanger #21: Internal error on command:"KTCC_T0"

anuargimenez, 2023-03-18:
Hi,

Thank you for creating such a valuable repository. Your hard work and dedication are appreciated.

I've been trying to use your code to control a Jubilee-style printer with two extruders. Almost all macros work fine, except for T1 and T0 (the two tools I have). I'm encountering the following error:

```
20:32:20
- DEBUG: _save_changes_timer_event:Exception: Internal error on command:"KTCC_T0"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
20:32:20
Internal error on command:"KTCC_T0"
20:32:20
Internal error on command:"KTCC_T0"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
20:32:20
- - TRACE: Saving state in logs.
20:32:14
Internal error on command:"T0"
20:32:14
Internal error on command:"_SET_ACTIVE_TOOL"
20:32:14
Internal error on command:"KTCC_T0"
20:32:14
Klipper state: Shutdown
20:32:14
- - TRACE: cmd_SelectTool: T0 - Not Virtual - Pickup
20:32:14
- - TRACE: increase_tool_statistics: After running for Tool: 0. Key: toolmounts_started is: 3
20:32:14
- - TRACE: increase_tool_statistics: Before running for Tool: 0. Key: toolmounts_started is: 2
20:32:14
- - TRACE: increase_tool_statistics: Running for Tool: 0. Provided to record tool stats while key: toolmounts_started and count: 1
20:32:14
- - TRACE: _set_tool_statistics: Tool: 0 provided to record tool stats while key: tracked_mount_start_time and value: 1678769407.61
20:32:13
- - TRACE: _set_tool_statistics:Running for Tool: 0 provided to record tool stats while key: tracked_mount_start_time and value: 1678769407.61
20:32:13
- - TRACE: track_mount_start: Running for Tool: 0.
20:32:13
- - TRACE: _set_tool_statistics: Tool: 0 provided to record tool stats while key: tracked_start_time_active and value: 1678769407.6
20:32:13
- - TRACE: _set_tool_statistics:Running for Tool: 0 provided to record tool stats while key: tracked_start_time_active and value: 1678769407.6
20:32:13
- - TRACE: track_active_heater_start: Running for Tool: 0.
20:32:13
- - TRACE: _set_tool_statistics_time_diff: Tool: 0 value after running: final_time_key: time_heater_standby=0, start_time_key: tracked_start_time_standby=0.
20:32:13
- - TRACE: track_standby_heater_end: Running for Tool: 0.
20:32:13
- - TRACE: .set_timer: T0; temp_type:0; duration:0.
20:32:13
- - TRACE: .set_timer: T0; temp_type:1; duration:0.
20:32:13
- - TRACE: This tool is_virtual is False.
20:32:13
- - TRACE: Current Tool is T-1.
20:32:13
- - TRACE: T0 Selected.
20:32:13
T0
```

This is my tools.cfg file:
[... 378 lines cut ...]
  SET_GCODE_VARIABLE MACRO=SAVE_ACCELERATION VARIABLE=max_accel VALUE={printer.toolhead.max_accel}

[gcode_macro RESTORE_ACCELERATION]
gcode:
  {% if printer['gcode_macro SAVE_ACCELERATION'].max_accel|int == 0 %}
    { action_respond_info("RESTORE_ACCELERATION: No acceleration saved.") }
  {% else %}
    M204 S{printer['gcode_macro SAVE_ACCELERATION'].max_accel}
  {% endif %}

[gcode_macro KTCC__DUMP__PRINT__STATS]
gcode:
  KTCC_DUMP_PRINT_STATS

[gcode_macro KTCC__INIT__PRINT__STATS]
gcode:
  KTCC_INIT_PRINT_STATS
```

Any help or guidance you could provide to resolve this issue would be greatly appreciated.

Thank you.

Sincerely,
Anuar R. Giménez

TypQxQ, 2023-03-18:
Happy you like it 😃 
Can you possibly zip the entire logfile, or a lot of it. The actual error is often higher up than the rows you supplied.
I see that you copied the config from my machine and while it's a good starting point it is a bit specific. Like for example this row:
SET_GCODE_VARIABLE MACRO=HOMING_STATUS VARIABLE=maxx VALUE=0         # Don't use the X-max endstop as EmergencyStop.

That deactivates my G-Code button but if you don't have one it will crash here. Error is probably listed elsewere in the log.

anuargimenez, 2023-03-20:
Hi,

Thank you for your prompt response.
I see... I didn't take into account what you mention. I don't know if the Klipper version is important either. 
This is all what is listed in the ktcc.log file: 

```
01:49:14 KlipperToolChangerCode Ready
01:49:15 - - TRACE: TOOL_UNLOCK running. 
01:49:24 KTCC Shutdown
01:49:27 KlipperToolChangerCode Ready
01:49:28 - - TRACE: TOOL_UNLOCK running. 
01:49:41 KTCC Shutdown
01:50:00 KlipperToolChangerCode Ready
01:50:02 - - TRACE: TOOL_UNLOCK running. 
01:51:47 KTCC Shutdown
02:01:29 KlipperToolChangerCode Ready
02:01:31 - - TRACE: TOOL_UNLOCK running. 
02:01:41 KTCC Shutdown
02:01:49 KlipperToolChangerCode Ready
02:01:51 - - TRACE: TOOL_UNLOCK running. 
02:01:59 KTCC Shutdown
02:06:43 KlipperToolChangerCode Ready
02:06:45 - - TRACE: TOOL_UNLOCK running. 
02:08:19 KTCC Shutdown
02:08:22 KlipperToolChangerCode Ready
02:08:23 - - TRACE: TOOL_UNLOCK running. 
02:09:57 KlipperToolChangerCode Ready
02:09:58 - - TRACE: TOOL_UNLOCK running. 
02:12:35 KlipperToolChangerCode Ready
02:12:37 - - TRACE: TOOL_UNLOCK running. 
02:16:46 KlipperToolChangerCode Ready
02:16:48 - - TRACE: TOOL_UNLOCK running. 
02:16:54 - - TRACE: ToolLock Unlocked.
02:16:54 - - TRACE: track_total_toolunlocks: Running.
02:16:54 ToolLock initialized unlocked
02:17:16 - - TRACE: Saving state in logs.
02:17:41 - - TRACE: TOOL_UNLOCK running. 
02:17:47 - - TRACE: ToolLock Unlocked.
02:17:47 - - TRACE: track_total_toolunlocks: Running.
02:17:54 - - TRACE: TOOL_LOCK running. 
02:17:56 - - TRACE: Saving state in logs.
02:17:57 - - TRACE: Tool Locked
02:17:57 - - TRACE: track_total_toollocks: Running.
02:18:06 - - TRACE: Saving state in logs.
02:18:10 - - TRACE: TOOL_UNLOCK running. 
02:18:16 - - TRACE: ToolLock Unlocked.
02:18:16 - - TRACE: track_total_toolunlocks: Running.
02:18:16 - - TRACE: Saving state in logs.
02:18:20 - - TRACE: T0 Selected.
02:18:20 - - TRACE: Current Tool is T-1.
02:18:20 - - TRACE: This tool is_virtual is False.
02:18:20 - - TRACE: <reactor.ReactorTimer instance at 0xb3182328>.set_timer: T0; temp_type:1; duration:0.
02:18:20 - - TRACE: <reactor.ReactorTimer instance at 0xb31823a0>.set_timer: T0; temp_type:0; duration:0.
02:18:20 - - TRACE: track_standby_heater_end: Running for Tool: 0.
02:18:20 - - TRACE: _set_tool_statistics_time_diff: Tool: 0 value after running: final_time_key: time_heater_standby=0, start_time_key: tracked_start_time_standby=0.
02:18:20 - - TRACE: track_active_heater_start: Running for Tool: 0.
02:18:20 - - TRACE: _set_tool_statistics:Running for Tool: 0 provided to record tool stats while key: tracked_start_time_active and value: 1678760300.2
02:18:20 - - TRACE: _set_tool_statistics: Tool: 0 provided to record tool stats while key: tracked_start_time_active and value: 1678760300.2
02:18:20 - - TRACE: track_mount_start: Running for Tool: 0.
02:18:20 - - TRACE: _set_tool_statistics:Running for Tool: 0 provided to record tool stats while key: tracked_mount_start_time and value: 1678760300.21
02:18:20 - - TRACE: _set_tool_statistics: Tool: 0 provided to record tool stats while key: tracked_mount_start_time and value: 1678760300.21
02:18:20 - - TRACE: increase_tool_statistics: Running for Tool: 0. Provided to record tool stats while key: toolmounts_started and count: 1
02:18:20 - - TRACE: increase_tool_statistics: Before running for Tool: 0. Key: toolmounts_started is: 1
02:18:20 - - TRACE: increase_tool_statistics: After running for Tool: 0. Key: toolmounts_started is: 2
02:18:20 - - TRACE: cmd_SelectTool: T0 - Not Virtual - Pickup
02:18:26 - - TRACE: Saving state in logs.
02:18:26 - DEBUG: _save_changes_timer_event:Exception: Internal error on command:"KTCC_T0"
         Once the underlying issue is corrected, use the
         "FIRMWARE_RESTART" command to reset the firmware, reload the
[... 576 lines cut ...]
05:02:31 - - TRACE: track_active_heater_start: Running for Tool: 1.
05:02:31 - - TRACE: _set_tool_statistics:Running for Tool: 1 provided to record tool stats while key: tracked_start_time_active and value: 1678770151.67
05:02:31 - - TRACE: _set_tool_statistics: Tool: 1 provided to record tool stats while key: tracked_start_time_active and value: 1678770151.67
05:02:31 - - TRACE: track_mount_start: Running for Tool: 1.
05:02:31 - - TRACE: _set_tool_statistics:Running for Tool: 1 provided to record tool stats while key: tracked_mount_start_time and value: 1678770151.68
05:02:31 - - TRACE: _set_tool_statistics: Tool: 1 provided to record tool stats while key: tracked_mount_start_time and value: 1678770151.68
05:02:31 - - TRACE: increase_tool_statistics: Running for Tool: 1. Provided to record tool stats while key: toolmounts_started and count: 1
05:02:31 - - TRACE: increase_tool_statistics: Before running for Tool: 1. Key: toolmounts_started is: 1
05:02:31 - - TRACE: increase_tool_statistics: After running for Tool: 1. Key: toolmounts_started is: 2
05:02:31 - - TRACE: cmd_SelectTool: T1 - Not Virtual - Pickup
05:02:40 - - TRACE: Saving state in logs.
05:02:40 - DEBUG: _save_changes_timer_event:Exception: Internal error on command:"KTCC_T1"
         Once the underlying issue is corrected, use the
         "FIRMWARE_RESTART" command to reset the firmware, reload the
         config, and restart the host software.
         Printer is shutdown
         
04:17:12 KlipperToolChangerCode Ready
04:17:13 - - TRACE: TOOL_UNLOCK running. 
04:17:20 - - TRACE: ToolLock Unlocked.
04:17:20 - - TRACE: track_total_toolunlocks: Running.
04:17:20 ToolLock initialized unlocked
04:17:22 - - TRACE: Saving state in logs.
```
Thank you for your time.

TypQxQ, 2023-04-01:
I will close this and hope it solved your problems 😄