TypQxQ/Klipper_ToolChanger #23: Offset calibration support

viesturz, 2023-04-10:
Hi, this is really cool. Wondering if you have plans to support automated nozzle offset calibration?

The conductive pin approach seems quite promising: 
https://youtu.be/ppcTm3NfDo0

viesturz, 2023-04-10:
That particular implementation seems to be a fork of this l: https://github.com/ankurv2k6/wp-daksh-toolchanger/blob/0ec6e951bd03cc12a109874a694ad207934ebb17/firmware/modules/Klipper_ToolChanger/probe_multi_axis.py

Would merge back in be a possibility?

TypQxQ, 2023-04-10:
I am plan ing to implement offset overrides or simmilar that can be saved from gcode macro. 
The implementation of a automatic probing, I leave to anyone willing. 
Me myself, I tried conductive probing for Z a couple years back and found it very unreliable because the smaklest, thinnest invisible coating of molten plastic on the nozzle acts as an insulator making the nozzle crash trough the probe into the bed. I had my conductive Z probe on a spring so it wouldn’t crash the bed and that’s how I came up with the KnobProbe. https://youtu.be/fshmEMWxBHY
for XY, I now use a microscope and TAMV. 

I only have to redo the calibration after a crash that bends the nozzle and luckily that doesn’t happen often.

closed 2023-04-10 by TypQxQ

viesturz, 2023-04-10:
I see, thanks. In my case I'm using Tap with a toolchanger, which makes crashes mostly impossible. And as you said this only needs to be done once in a while, so can clean it all up beforehand.

The above seems to originate from
https://github.com/ben5459/Klipper_ToolChanger 
That has quite some activity last year.
Might be worth checking out if there is something to merge back.

TypQxQ, 2023-04-11:
You can also check the KnobProbe code. Will check if I can upmerge changes :)/Andrei10 apr.

https://github.com/ben5459/Klipper_ToolChanger