TypQxQ/Klipper_ToolChanger #26: Heater macros don't work

Bimbovadasz, 2023-05-28:
Hi!
Thank you for creating this plugin. It would make life so much easier but unfortunately, I can't make it work properly.
I have 2 tools for now, but I don't have any changing movement yet, I just want to verify that all the basic functions work.  For some reason the heating macros don't work for me. I replaced the originals with the provided M104 and M109 macros, but they do nothing when they are called. Also SET_TOOL_TEMPERATURE and TEMPERATURE_WAIT_WITH_TOLERANCE doesn't work either. 
If I call SET_TOOL_TEMPERATURE TOOL=0 ACTV_TMP=200 and then SET_TOOL_TEMPERATURE TOOL=0 it shows that it changed the temp value in code, but the heater won't turn on. 

[image]

I'm not the best in programming, so maybe I missed a step in the configuration. I upload my configs.
tc_test.cfg is for the second toolhead setup, macro_remap.cfg is for the new macros and ktcc.cfg is for the plugin's config.
Could you please check if I've done the configs right? 
[klipper_config_20230528.zip](https://github.com/TypQxQ/Klipper_ToolChanger/files/11584651/klipper_config_20230528.zip)

Thank you!

TypQxQ, 2023-05-30:
Hi!
Can you try adding a "CHNG_STATE =2" parameter to make the tool active?

Bimbovadasz, 2023-05-30:
Oh, I see
Yes that worked! 
So for example at the start of the print I set the active and standby temp, and then I just have to change the states as the tools change.
This is great, it just wasn't clear how to use the macro.
Thank you for the help!

closed 2023-05-30 by Bimbovadasz