TypQxQ/Klipper_ToolChanger #28: Start Macro on Extruder 2 while printing with Extruder 1

AlexBollmann, 2023-08-10:
Hi,

is it possible to have an extruder in parking position already heating and purging, while the printer is busy printing with another extruder? That would save a lot of time, because the tool change could be really quick.

TypQxQ, 2023-08-10:
It would but the problem is not knowing a toolchange will happen beforehand. I use the timers and have the tools waiting in standby at 175*C for 10minutes after a toolchange from them. When toolchanging, the first it does is putting nexttool in active mode and heats up before beginning to park the current tool. If its in standby at 175*C my Revo is done heating up when it is ready to undock so I have no problems. If the slicer wold issue a prepare tool command for example 30s before a toolchange it would be easy but Klipper has no lookahead for commands and cannot know when next toolchange is going to happen. /Andrei10 aug.

AlexBollmann, 2023-08-16:
That's a good point which is clearly in the scope of the slicer. I was afraid klipper permits the extrusion if the tool is not "selected". If that's not the case, I should look into the slicer.

TypQxQ, 2023-08-16:
Great.  Thank you.

closed 2023-08-16 by TypQxQ