TypQxQ/Klipper_ToolChanger #29: Tool changing macro setup issue

Bimbovadasz, 2023-08-18:
Hi!
I'm trying to get physical tool changing work, but I keep getting errors. I am building a tool changer similar to the E3D one, but with an electromagnet instead of the stepper.
I'm using the code for testing that I found in /config/example_simple/tools.cfg, but simplified it a bit for my configuration. 
```
[gcode_macro TOOL_PICKUP]
gcode:
  M568 P{myself.name} A2                                               # Put tool heater in Active mode
  SAVE_GCODE_STATE NAME=TOOL_PICKUP                                    # Save GCODE state. Will be restored at the end of pickup code
  G90
  G0 X{myself.zone[0]} Y{myself.zone[1]} F18000                        # Fast move Y in front of parking spot.
  G0 Y{myself.park[1]} F1800                                           # Slow Move to the pickup position for tool.
  M400                                                                 # Wait for all moves to finish.
  TOOL_LOCK                                                            # Lock the tool.
  SET_GCODE_OFFSET Z={myself.offset[2]} MOVE=1                         # Set and move the Z offset. Avoid crashing into bed when moving out later.
  {% if myself.extruder|default("none")|lower !="none" %}              # If the tool has an extruder:
    TEMPERATURE_WAIT_WITH_TOLERANCE TOOL={myself.name}                 # Wait for tool to reach target temperature.
  {% endif %}                                                          
  G0 X{myself.zone[0]} F1800                                           # Slow Move to the zone position for tool.
  M400                                                                 # Wait for current moves to finish.
  RESTORE_GCODE_STATE NAME=TOOL_PICKUP MOVE=0                          # Restore GCODE state. Was saved at thebegining of SUB_TOOL_PICKUP_START. Move fast to last location.
                                                                       # Set the toolhead offsets. Z is set and moved before any moves in SUB_TOOL_PICKUP_START. Needs to be after any RESTORE_GCODE_STATE!
  SET_GCODE_OFFSET X={myself.offset[0]} Y={myself.offset[1]} Z={myself.offset[2]} MOVE=0  # Set X and Y offsets,
```
But when I call the macro I get this error:
Error evaluating 'gcode_macro TOOL_PICKUP:gcode': UndefinedError: 'myself' is undefined

Am I missing something?

TypQxQ, 2023-08-18:
The macro is called by the Python code and passes those parameters. /Andrei18 aug.

Bimbovadasz, 2023-08-19:
I see that, but it doesn't work. 
I get the error, and I don't know what to do. 
Can you help me?

TypQxQ, 2023-08-20:
I see that I too simplified it a bit much 😄 
I changed the example code for the SUB_TOOL_DROPOFF.

It is called with inside the tool  configuration like this:
dropoff_gcode:
  SUB_TOOL_DROPOFF T={myself.name}

and it needs to reasign the myself in the begining of code with:
{%set myself = printer['tool '~params.T]%}

Current backups of my working configuration can be used as inspiration:
https://github.com/TypQxQ/DuetBackup/tree/main/qTC-Klipper

Bimbovadasz, 2023-08-29:
Okay, it worked, thank you! 😄

closed 2023-08-29 by Bimbovadasz